ESP32-Musings/OTA_update_STM32_using_ESP32 #3: How to build and flash using esp-idf command prompt

AliHassanEB, 2020-11-14:
Hi I have been using ESP-IDF command prompt and couldn't build and flash the code. can you explain or give some steps to upload. I am new to esp-idf

laukik-hase, 2020-11-15:
@AliHassanEB For getting started with ESP-IDF, you can refer their [official documentation](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html). This repository is compatible with ESP-IDF v4.1, so make sure you have that version installed.

AliHassanEB, 2020-11-16:
Thank you for replying. I have installed esp-idf by using ESP-IDF Tools Installer. Then opened  ESP-IDF Command Prompt (cmd.exe) and wrote the command "cd E:\Data\OTA_update_STM32_using_ESP32\file_serving_stm" . After changing the folder I wrote "idf.py build" but after some compilation, an error popped up "fatal error: stm_flash.h: No such file or directory"

AliHassanEB, 2020-11-16:
Somehow I think I successfully build and flash the code by rewriting the CMakeLists.txt line "set(EXTRA_COMPONENT_DIRS $ENV{HOME}/projects/OTA_update_STM32_using_ESP32/components" to "set(EXTRA_COMPONENT_DIRS E:/Data/projects/OTA_update_STM32_using_ESP32/components)" and placed my OTA_update_STM32_using_ESP32 folder in a new folder called projects

AliHassanEB, 2020-11-16:
On the webpage, it is saying "Directory does not exist" and on the monitor.
 4179) SPIFFS: spiffs partition could not be found.
(4179) : stm_pro_mode (initSPIFFS:45) Failed to find SPIFFS partition.

laukik-hase, 2020-11-17:
@AliHassanEB First try running all the basic examples on your ESP32 Module to verify the working of all peripherals. Following is a list:

1. [SPIFFS Basic](https://github.com/espressif/esp-idf/tree/release/v4.1/examples/storage/spiffs)
2. [UART Async Tasks](https://github.com/espressif/esp-idf/tree/release/v4.1/examples/peripherals/uart/uart_async_rxtxtasks)
3. [HTTP File Server](https://github.com/espressif/esp-idf/tree/release/v4.1/examples/protocols/http_server/file_serving)

After verifying the working of these examples, you can then move on to flashing our code. For any trouble regarding our code, you can always raise an issue here, but for your ESP-IDF related doubts, you should go to their [forum](https://www.esp32.com/viewforum.php?f=23&sid=14b1a53eae58c55d470fa64ca58b55aa).

closed 2020-11-17 by laukik-hase