ESP32-Musings/OTA_update_STM32_using_ESP32 #4: Serial Timeout ERROR

muhambykar, 2020-11-28:
Hi for all, I am trying to use this example, but I couldnt it successfully. I got Serial timeout error.( esp32 and stm32 could not talk successfully. ) I have tried everything, powered up stm32 and esp32 with different sources, erased stm32 flash memory with STMFlashLoader, changed up baudrate, or buttons and still got error. What could be the reason of this? 
Thank you...

D (337256) : stm_flash (flashSTM:74) File name: /spiffs/blink.bin
I (337256) : stm_pro_mode (initGPIO:76) GPIO Initialized
I (337256) : stm_flash (flashSTM:83) Writing STM32 Memory
I (337256) : stm_flash (writeTask:7) Write Task
I (337256) : stm_pro_mode (resetSTM:81) Starting RESET Procedure
I (337856) : stm_pro_mode (resetSTM:88) Finished RESET Procedure
I (337856) : stm_pro_mode (cmdSync:114) SYNC
E (337896) : stm_pro_mode (sendBytes:231) Serial Timeout
I (337896) : stm_pro_mode (cmdGet:123) GET
E (337936) : stm_pro_mode (sendBytes:231) Serial Timeout
I (337936) : stm_pro_mode (cmdVersion:132) GET VERSION & READ PROTECTION STATUS
E (337976) : stm_pro_mode (sendBytes:231) Serial Timeout
I (337976) : stm_pro_mode (cmdId:141) CHECK ID
E (338016) : stm_pro_mode (sendBytes:231) Serial Timeout
I (338016) : stm_pro_mode (cmdErase:149) ERASE MEMORY
E (338056) : stm_pro_mode (sendBytes:231) Serial Timeout
I (338056) : stm_pro_mode (cmdExtErase:166) EXTENDED ERASE MEMORY
E (338096) : stm_pro_mode (sendBytes:231) Serial Timeout
I (338106) : stm_flash (writeTask:19) Writing block: 1
I (338106) : stm_pro_mode (flashPage:279) Flashing Page
I (338106) : stm_pro_mode (cmdWrite:183) WRITE MEMORY
E (338146) : stm_pro_mode (sendBytes:231) Serial Timeout
E (338186) : stm_pro_mode (sendBytes:231) Serial Timeout
E (338496) : stm_pro_mode (flashPage:318) Serial Timeout
I (338496) : stm_flash (flashSTM:92) STM32 Flashed Successfully!!!
I (338496) : stm_flash (flashSTM:95) Ending Connection
I (338496) : stm_pro_mode (resetSTM:81) Starting RESET Procedure
I (339096) : stm_pro_mode (resetSTM:88) Finished RESET Procedure
I (339096) : stm_pro_mode (endConn:98) Ending Connection
I (339096) : stm_flash (flashSTM:98) Closing file
E (339096) : FILE_SERVER_HTTP (flash_post_handler:445) Target flashing failed

muhambykar, 2020-11-30:
Is there anyone got same error?

laukik-hase, 2020-11-30:
Hello, @muhambykar

Can you check the GPIOs that your are using as Rx and Tx?

muhambykar, 2020-11-30:
I have checked rx tx pins, I can write and erase with these pins on FlashLoaderDemo.exe but I couldnt it with esp32. I have tried directly Boot0 to HIGH, and I have also tried with  #define BOOT0_PIN (GPIO_NUM_21)  But got some error log. Is there anything that I can check?

muhambykar, 2020-12-02:
Is there anybody got same error?

laukik-hase, 2020-12-08:
Hello, @muhambykar
Sorry for the late reply.

1) Test this example for UART - [UART Async Tasks](https://github.com/espressif/esp-idf/tree/release/v4.1/examples/peripherals/uart/uart_async_rxtxtasks)
2) Verify whether your STM32 is in BOOT mode before flashing.
3) Try resetting your STM32 using the RESET switch on your board, and the running the example. (Time it appropriately)

LetsControltheController, 2020-12-09:
Hello there, I found the solution. on ESP-IDF v4.0.x , 
    uart_set_rts(UART_CONTROLLER, 1);     
this function changes the uart configs, and got trouble about it.  After make comment it, problem is solved. This line is on components/stm_pro_mode/stm_pro_mode.c  Line: 19

I have listened the received packets from UART, then I figure out it and it works. I recommend it.

laukik-hase, 2020-12-12:
> Hello there, I found the solution. on ESP-IDF v4.0.x ,
> uart_set_rts(UART_CONTROLLER, 1);
> this function changes the uart configs, and got trouble about it. After make comment it, problem is solved. [...]

This function was inserted as the STM32 flashing protocol works on USART, not UART. We have tested with two different version of the cjip (STM32F103 and STM32F072) and got successful results. Can you guys tell which STM32 chip are you using?

LetsControltheController, 2020-12-14:
Its not about STM32, its about ESP32. When we configure RTS manually iwth this function, ESP32 could not read data on Rx pin. So if we leave it with automatic configuration, it works.

laukik-hase, 2020-12-21:
@muhambykar I am closing the issue for now. Reopen it if you have any updates regarding the problem.

madhavee77, 2021-09-01:
Hey I am getting the same error and tried everything mentioned above. Was someone able to solve it?

akym, 2021-11-17:
I have the same issue, i tried all the solutions mentioned in this issue and then some more. Cant seem to point out why esp32 cant receive anything, even tried the uart_set_rts solution mentioned by @LetsControltheController . The UART example works but only when talking with STM it doesn't work. Any thoughts on the problem?

laukik-hase, 2021-11-19:
Hello,

Sorry for the late response. I will try reproducing this on my side and revert back to you.

@akym and @madhavee77 Could you guys specify the following:
1. ESP-IDF Version (use `git describe --tags` to find out)
2. STM32 MCU (e.g. `STM32F103` (Blue-pill), `STM32F401` (Black-pill), `STM32F072`, etc.)

akym, 2021-11-19:
Hi, 
Thank you for the quick response. Here are the things you requested:
1.ESP-IDF 4.3
2. STM43F4

I probed Tx,Rx of the STM32 with oscilloscope and it is transmitting some data on the Tx but ESP32 doesnt read any of that data. I have tried running the UART example as you suggested and that one works fine on UART0 and UART2. For my implementation I am using UART2 to push firmware to the STM.
cd 
[image]

Any help will be appreciated.

My code for reference:

[stmOTA.zip](https://github.com/ESP32-Musings/OTA_update_STM32_using_ESP32/files/7569846/stmOTA.zip)

laukik-hase, 2021-11-29:
> Hello there, I found the solution. on ESP-IDF v4.0.x , uart_set_rts(UART_CONTROLLER, 1); this function changes the uart configs, and got trouble about it. After make comment it, problem is solved. [...]

Hello, @akym

Sorry for the late update. I was able to reproduce this issue on ESP-IDF branch `release/v4.3` with an `STM32F103`. But, this fix (suggested by @LetsControltheController) solved the issue for me. I will try reproducing with an `STM32F401` and revert back to you.

Are you resetting the board before flashing? (if you are using the on-board BOOT and RESET buttons, you will need to time it appropriately)

akym, 2021-12-01:
Thank you for the effort.

I tried the set_rts but it did not work. 

I am resetting the board with flipping ESP's GPIO pin to NRST of STM32F4, not manually. My code is attached in the original question, but it is almost the same as this repo.

laukik-hase, 2021-12-04:
Hello, @akym

laukik-hase, 2021-12-04:
Hello, @akym 

I tried the example again on an [`STM32F103C8`](https://stm32-base.org/boards/STM32F103C8T6-Blue-Pill) with `uart_set_rts(...)` disabled and the connections mentioned in the README, but it executed successfully.

I also had an [`STM32F401xC`](https://stm32-base.org/boards/STM32F401CCU6-WeAct-Black-Pill-V1.2.html) lying around so I gave it a try, but the example failed with the `Serial Timeout` error.

laukik-hase, 2021-12-04:
Hello, @akym 

I tried the example again on an [`STM32F103C8`](https://stm32-base.org/boards/STM32F103C8T6-Blue-Pill) with `uart_set_rts(...)` disabled and the connections mentioned in the README, but it executed successfully.

I also had an [`STM32F401xC`](https://stm32-base.org/boards/STM32F401CCU6-WeAct-Black-Pill-V1.2.html) lying around so I gave it a try on that, but the example failed with the `Serial Timeout` error.

1. Could you try the connections as mentioned in this [article](http://stm32f4-discovery.net/2014/09/program-stm32f4-with-uart/)? As you are working with a Discovery board, you could be needed to change the pins (`PA9` and `PA10`) for USART communication.
2. My `STM32F401xC` board went straight into the `DFU  Bootloader` mode on setting the `BOOT0`, `BOOT1` and `nRST` pins as required and was able to flash through the `USB` mode with `STM32CubeProgrammer` after that. I am still investigating how to enter the `USART Bootloader` mode on these boards, will revert back to you if I find out more.

akym, 2022-01-04:
Hello @laukik-hase ,
Happy new year and sorry for the late reply. 

1. I did try the connection suggested in the link you sent and other ports from the post as well.
2. I tried flashing STM32 over USART with and external FTDI, it works flawlessly with 'STM Flash Loader Demonstrator'

I also tried following the AN3155 ('USART protocol used in the STM32 bootloader') from STM to see where things might not be working. maybe you can try having a look. That is when I probed the TX & RX pins with osci to see if there is anything.

madhavee77, 2022-02-01:
Hey everyone, sorry for the late reply.

I was able to resolve this issue on my end. I first tested if the STM32 is working or not by putting it in bootloader mode and started sending the setup bytes manually to it at 115200 baud rate, even parity, using FTDI. Whenever I sent something, I was receiving the Acknowledgement byte which is 0x79. Which proved there was nothing wrong with STM32.

Then I tried testing a basic UART example of ESP32 with the same settings as in this code (even including uart_set_rts(UART_CONTROLLER, 1);). That's when I found that the issue was because of this function in the stm_pro_mode.c line 244.

waitForSerialData();

Most probably the function uart_get_buffered_data_len() is the issue and is not reading anything on the receiving line whenever STM32 sends something back.

This function is not able to read the incoming data, maybe its supporting libraries are not available on the latest version or whatever is the reason I don't know but this was the issue on my end. I removed it and made it dynamic.

This is how I edited the function which was using the waitForSerialData() at line 206:

int sendBytes(const char *bytes, int count, int resp)
{
    sendData(TAG_STM_PRO, bytes, count);
  //  int length = waitForSerialData(resp, SERIAL_TIMEOUT);
    uint8_t* data = (uint8_t*)malloc(resp * sizeof(uint8_t*));

    int rxBytes = uart_read_bytes(UART_CONTROLLER, data, UART_BUF_SIZE, 1000 / portTICK_RATE_MS);
        
    if (rxBytes > 0 ) 
    {
        if(rxBytes == resp && data[0] == ACK)
        {
            data[rxBytes] = 0;
            logI(TAG_STM_PRO, "%s", "Sync Success");
           // ESP_LOG_BUFFER_HEXDUMP(TAG_STM_PRO, data, rxBytes, ESP_LOG_INFO);
            return 1;
        }
        else
        {
            logE(TAG_STM_PRO, "%s", "Sync Failure");
            return 0;
        }
    }
    else
    {
        logE(TAG_STM_PRO, "%s", "Serial Timeout");
        return 0;
    }
    free(data);
    return 0;
}

Similar changes you may need to make wherever this function is being used. 

I hope this information helps you.

Thank you

madhavee77, 2022-02-01:
Hey everyone, sorry for the late reply.

I was able to resolve this issue on my end. I first tested if the STM32 is working or not by putting it in bootloader mode and started sending the setup bytes manually to it at 115200 baud rate, even parity, using FTDI. Whenever I sent something, I was receiving the Acknowledgement byte which is 0x79. Which proved there was nothing wrong with STM32.

Then I tried testing a basic UART example of ESP32 with the same settings as in this code (even including uart_set_rts(UART_CONTROLLER, 1);). That's when I found that the issue was because of this function in the stm_pro_mode.c line 244.

waitForSerialData();

Most probably the function uart_get_buffered_data_len() is the issue and is not reading anything on the receiving line whenever STM32 sends something back.

This function is not able to read the incoming data, maybe its supporting libraries are not available on the latest version or whatever is the reason I don't know but this was the issue on my end. I removed it and made it dynamic.

This is how I edited the function which was using the waitForSerialData() at line 206:

int sendBytes(const char *bytes, int count, int resp)
{
sendData(TAG_STM_PRO, bytes, count);

uint8_t* data = (uint8_t*)malloc(resp * sizeof(uint8_t*));

int rxBytes = uart_read_bytes(UART_CONTROLLER, data, UART_BUF_SIZE, 1000 / portTICK_RATE_MS);
    
if (rxBytes > 0 ) 
{
    if(rxBytes == resp && data[0] == ACK)
    {
        data[rxBytes] = 0;
        logI(TAG_STM_PRO, "%s", "Sync Success");
       // ESP_LOG_BUFFER_HEXDUMP(TAG_STM_PRO, data, rxBytes, ESP_LOG_INFO);
        return 1;
    }
    else
    {
        logE(TAG_STM_PRO, "%s", "Sync Failure");
        return 0;
    }
}
else
{
    logE(TAG_STM_PRO, "%s", "Serial Timeout");
    return 0;
}
free(data);
return 0;
}
`

Similar changes you may need to make wherever this function is being used.

I hope this information helps you.

Thank you

rickie95, 2022-04-08:
Just a tip, my 2 cents: I've encountered this issue on a NUCLEO **STM32G431**. In my case the ESP board could not reset the STM and this led to a SYNC error just as described in this thread. **Even manually grounding the NRST** pin the STM refused to reset. I end up turning off and on the NUCLEO prior to the flash procedure, and then successfully flashed the firmware.

adirsingh96, 2022-08-17:
Do we have to install some bootloader in STM32 MCU(I am using F103RB) to flash bin file using UART? I have tried every solution mentioned in the comments and still getting sync error / serial Timeout Error

adirsingh96, 2022-08-17:
> Do we have to install some bootloader in STM32 MCU(I am using F103RB) to flash bin file using UART? I have tried every solution mentioned in the comments and still getting sync error / serial Timeout Error 

Solved this, I was using the wrong UART port of STM32 and boot configuration was not right

bjesuslopezg, 2023-02-04:
I made it work by redefining the macro of SERIAL_TIMEOUT located in the stm_pro_mode.h to an extremely large value:
#define SERIAL_TIMEOUT 150000000

It was previously defined as 5000. My hypothesis was that since the problem is the timing out of the request, then let's extend the maximum time of it and see what happens. The result was that it works, but it takes like 10 seconds to upload the code to the STM32.

I'm using STM32F407 and ESP32-WROVER-B

pinkymaxou, 2023-03-14:
I submitted a fix: https://github.com/ESP32-Musings/OTA_update_STM32_using_ESP32/pull/16

pinkymaxou, 2023-03-14:
> I made it work by redefining the macro of SERIAL_TIMEOUT located in the stm_pro_mode.h to an extremely large value: #define SERIAL_TIMEOUT 150000000
>
> It was previously defined as 5000. My hypothesis [...]

That's because the SerialWait function doesn't work on current master, put a gigantic number make the while() run longer despite the not working delay. My fix should improve your situation.

Uvamosk, 2024-05-22:
> @akym Long shot but did you ever get this working with the STM32F407 Discovery Board? I am attempting to do the same thing and I have a interesting result. It's partially working.
> 
> It gets to ERASE MEMORY and has Sync Failure but then does EXTENDED ERASE MEMORY and has Sync Success. My boards blinky light program seems to be erased as the board stops blinking the leds.
> 
> Can I assume that the Serial Timeout after the (Extended Erase Memory) Sync Success indicates I need to increase the @bjesuslopezg Serial_Timeout / apply the fix that @pinkymaxou these two suggest
> 
> Here is the console output: I (6667) : stm_pro_mode (initGPIO:75) GPIO Initialized I (6667) : stm_flash (flashSTM:83) Writing STM32 Memory I (6667) : stm_flash (writeTask:7) Write Task I (6677) : stm_pro_mode (resetSTM:80) Starting RESET Procedure I (7277) : stm_pro_mode (resetSTM:87) Finished RESET Procedure I (7277) : stm_pro_mode (cmdSync:113) SYNC I (7277) : stm_pro_mode (sendBytes:218) Sync Success I (7277) : stm_pro_mode (cmdGet:122) GET I (7287) : stm_pro_mode (sendBytes:218) Sync Success I (7287) : stm_pro_mode (cmdVersion:131) GET VERSION & READ PROTECTION STATUS I (7297) : stm_pro_mode (sendBytes:218) Sync Success I (7307) : stm_pro_mode (cmdId:140) CHECK ID I (7307) : stm_pro_mode (sendBytes:218) Sync Success I (7317) : stm_pro_mode (cmdErase:148) ERASE MEMORY E (7317) : stm_pro_mode (sendBytes:224) Sync Failure I (7327) : stm_pro_mode (cmdExtErase:165) EXTENDED ERASE MEMORY I (7337) : stm_pro_mode (sendBytes:218) Sync Success E (7407) : stm_pro_mode (sendBytes:230) Serial Timeout I (7407) : stm_flash (writeTask:19) Writing block: 1 I (7407) : stm_pro_mode (flashPage:278) Flashing Page I (7417) : stm_pro_mode (cmdWrite:182) WRITE MEMORY E (7487) : stm_pro_mode (sendBytes:230) Serial Timeout E (7557) : stm_pro_mode (sendBytes:230) Serial Timeout E (7647) : stm_pro_mode (flashPage:317) Serial Timeout I (7647) : stm_flash (flashSTM:94) Ending Connection I (7647) : stm_pro_mode (resetSTM:80) Starting RESET Procedure I (8257) : stm_pro_mode (resetSTM:87) Finished RESET Procedure I (8257) : stm_pro_mode (endConn:97) Ending Connection I (8257) : stm_flash (flashSTM:97) Closing file E (8257) : FILE_SERVER_HTTP (flash_post_handler:444) Target flashing failed I (8357) FILE_SERVER_HTTP: Found file : STM32_Spin_Blinky.bin (30928 bytes)

For anyone who finds this and is doing a STM32F407 Discovery board
    I change sdkconfig from 100 to 1000hz ticks  
    stm_pro_mode.h
        Change pins at LINE: 39 and LINE: 47
        Change SERIAL_TIMEOUT 20000 (I tried 15000 and it was getting timeouts)
   stm_pro_mode.c
        Changed LINE : 263 to vTaskDelay(1)

Used these pins
 PB10 -> ESP32 RX
 PB11 -> ESP32 TX
 BOOT0 -> GPIO_25
 BOOT1 (PB2) -> Directly to ground
 nRST -> GPIO_27
 
 Worked perfectly for me.

Sudeep131084, 2024-08-30:
int sendBytes(const char *bytes, int count, int resp)
{
    sendData(TAG_STM_PRO, bytes, count);
	uart_wait_tx_done(UART_CONTROLLER, pdMS_TO_TICKS(1000));
    int length = waitForSerialData(resp, SERIAL_TIMEOUT);

    if (length > 0)
    {
        uint8_t data[length];
       // const int rxBytes = uart_read_bytes(UART_CONTROLLER, data, length, 1000 / portTICK_RATE_MS);
		const int rxBytes = uart_read_bytes(UART_CONTROLLER, data, length, 1000 / portTICK_PERIOD_MS);
        //logI(TAG_STM_PRO, "Response : %02x", data[0]);
        if (rxBytes > 0 && data[0] == ACK)
        {
            logI(TAG_STM_PRO, "%s", "Sync Success");
            // ESP_LOG_BUFFER_HEXDUMP("SYNC", data, rxBytes, ESP_LOG_DEBUG);
            return 1;
        }
        else
        {
            logE(TAG_STM_PRO, "%s", "Sync Failure");
            return 0;
        }
    }
    else
    {
        logE(TAG_STM_PRO, "%s", "Serial Timeout");
        return 0;
    }

    return 0;
}
int waitForSerialData(int dataCount, int timeout)
{
    int timer = 0;
    int length = 0;
    while (timer < timeout)
    {
        uart_get_buffered_data_len(UART_CONTROLLER, (size_t *)&length);
        if (length >= dataCount)
        {
            return length;
        }
		//vTaskDelay(pdMS_TO_TICKS(1000)); 
        vTaskDelay(10 / portTICK_PERIOD_MS);
        timer++;
    }
    return 0;
}

After some changes in these function my code is working for STM32F401RE necleo board.