ESP32-Musings/OTA_update_STM32_using_ESP32 #7: running the project in vscode

Benedito821, 2021-07-20:
Hello! thanks for the project. How to compile it in vscode if we have here two directories  _esp_stm_flash_ and file_serving_stm and each of them has its own main file?

laukik-hase, 2021-07-21:
Hello, @Benedito821. Glad to see the project helped you.

Both the directories are different examples, you can build them according to your purpose. Refer [this](https://github.com/laukik-hase/OTA_update_STM32_using_ESP32#code) section for more info.

Benedito821, 2021-07-21:
can I use this example to flash the board esp itself(not the stm32)?For instance if I don't switch the jumper

Benedito821, 2021-07-21:
I am facing a very annoying problem now: the compiler can't find the file stm_flash.h although it is in the folder components. I tried to add the path in .json file but nothing actually changed. Do you have any suggestions on how to solve this?

{
    "configurations": [
        {
            "name": "Linux",
            "includePath": [
                "${workspaceFolder}/**",
                **"/home/.../.../OTA_update_STM32_using_ESP32/components/"**
            ],
            "defines": [],
            "compilerPath": "/usr/bin/gcc",
            "cStandard": "gnu17",
            "cppStandard": "gnu++14",
            "intelliSenseMode": "linux-gcc-x64",
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
}

Benedito821, 2021-07-25:
can I use this project to flash not stm but esp itself(if don't switch the jumper)?

Benedito821, 2021-07-28:
how could I solve this issue ?
E (5613) SPIFFS: spiffs partition could not be found
E (5613) : stm_pro_mode (initSPIFFS:45) Failed to find SPIFFS partition

laukik-hase, 2021-08-01:
Hello, @Benedito821
You cannot use this project to flash an ESP32 using another. You can take a look at [this](https://github.com/espressif/esp-serial-flasher) project for more information and play with it according to your use case.

Regarding the SPIFFS issue, can you run [this](https://github.com/espressif/esp-idf/tree/master/examples/storage/spiffs) example and check if you get the same problem? Also, check your flash size and modify the [partition table](https://github.com/laukik-hase/OTA_update_STM32_using_ESP32/blob/master/file_serving_stm/partitions_example.csv) accordingly. (I had a 4MB flash model, you might have a 2MB one).

Benedito821, 2021-08-02:
thanks for your reply! I meant not one esp flashing another, but flashing that esp that was supposed to program the stm(the esp itself) as I have only one esp.

Benedito821, 2021-08-02:
I runned the SPIFFS example and it worked as expected.
Regarding the SPIFFS: I have a 8MB maximum flash size but I currently use 4MB. How can I change the partition table? In menunconfig there is not option for this. Below my partition summary:
# Name, Type, SubType,       Offset, Size, Flags
nvs,         data,         nvs,     0x9000,  24K,
phy_init,  data,         phy,      0xf000,    4K,
factory,     app,    factory,   0x10000,   1M,

I see that you have an additional line : storage,  data, spiffs,  ,        960K,

laukik-hase, 2021-08-03:
Hello, @Benedito821 
Here is the [csv](https://github.com/laukik-hase/OTA_update_STM32_using_ESP32/blob/master/file_serving_stm/partitions_example.csv) from the example.  Reduce the SPIFFS partition size if you have 2MB flash. Check [this](https://github.com/espressif/esptool#read-spi-flash-id-flash_id) out for finding that.

Benedito821, 2021-08-03:
Hello, @laukik-hase 
I have a maximum 8MB flash, but I the definition is set 4MB (I think no problem with this). I've seen your [csv](https://github.com/laukik-hase/OTA_update_STM32_using_ESP32/blob/master/file_serving_stm/partitions_example.csv) file. My problem now is to edit my partition table. Could you tell me please how you edited your partition table, if I am not disturbing?

Benedito821, 2021-08-03:
> I runned the SPIFFS example and it worked as expected.Could you please give a hint on how to modify my partition table? I found an article [Partition tables](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html) of esp but I just named my csv file and I don't know what to do further.
> Here is my partition table:
> 
> Name, Type, SubType, Offset, Size, Flags
> nvs,data,nvs,0x9000,24K,
> phy_init,data,phy,0xf000,4K,
> factory,app,factory,0x10000,1M,

Benedito821, 2021-08-05:
Hello, @laukik-hase 
I have just edited my partition table and the SPIFFS error has gone. Almost there. But now he is complaining about my .bin file size(it is 172.016 bytes). Where can I increase the supported size?

_W (35591) httpd_parse: parse_block: request URI/header too long
W (35591) httpd_txrx: httpd_resp_send_err: 431 Request Header Fields Too Large - Header fields are too long for server to interpret_

Benedito821, 2021-08-05:
Everything has worked but one: the flash failed(because I didin't connect a stm32 board). So, could you please answer this quaestion, once it is probably the reason of the error. Thanks for the help so far!

> thanks for your reply! I meant not one esp flashing another, but flashing that esp that was supposed to program the stm(the esp itself) as I have only one esp.

laukik-hase, 2021-08-09:
Hello, @Benedito821

You can flash the ESP using OTA. There are examples in ESP-IDF and also many applications from other users.

Benedito821, 2021-08-09:
I honestly saw the esp OTA example. But the problem is that it doesn't have a file server as in your project, so the bin file upload is a bit awkward. Do you know an OTA example that has a file server?

laukik-hase, 2021-08-15:
Hello, @Benedito821 
What are you looking for exactly? Do you want to flash ESP32 instead of STM32 after uploading the file to the file server?

Benedito821, 2021-08-15:
Yes!!!

laukik-hase, 2021-08-17:
Hello, @Benedito821 
Is this what you are looking for: https://github.com/versamodule/ESP32-OTA-Webserver

I am closing this issue now, if you face any other problems with our project, please feel free to raise another issue.

closed 2021-08-17 by laukik-hase

Benedito821, 2021-08-20:
hello! the link sttoped working! error 404